uleroboticsgroup/yasmin #3: Practical example

fzoric8, 2022-08-18:
Hi, 

I'm checking yasmin as simple state manager for ROS2 purposes. 
Besides `simple_demo.py` there's no clear example on how to use it. 

I'm wondering if it would be possible to implement following in `State` class. 
 * `publish/subscribe` to ROS topic 
 * `call` ROS service 

I've figured out `demo` script, however, I'm not sure how to properly use current `State` implementation. 

I can initialize ROS node with certain publishers/subscribers. However, if i pass certain values I
want to monitor during state execution (robot pose), they remain unchanged during state execution. 
Execute function is used, however, state runs alongside ROS node and it's not possible to 
continuously update robot pose value in State class. 

You can see modified demo [here](https://github.com/fzoric8/yasmin/blob/main/yasmin_demo/yasmin_demo/yasmin_demo.py). 

My question is following: 
* Is there any example of a state that monitors certain ROS topic, and is it possible to use information 
from ROS topic in State without initializing subscriber in State class directly. 

Thank you for your answer.

fzoric8, 2022-09-06:
Inactive repo, don't use it.

mgonzs13, 2022-11-01:
Hi @fzoric8, sorry for the delay.
I've just created a MonitorState for [Python](https://github.com/uleroboticsgroup/yasmin/blob/main/yasmin_ros/yasmin_ros/monitor_state.py) and [C++](https://github.com/uleroboticsgroup/yasmin/blob/main/yasmin_ros/include/yasmin_ros/monitor_state.hpp).
[Here](https://github.com/uleroboticsgroup/yasmin/blob/main/yasmin_demo/yasmin_demo/monitor_demo.py) you can see an example of how to use this new state to monitor the odom topic.

closed 2023-01-03 by mgonzs13