uleroboticsgroup/yasmin #16: Disabling visualization of nested FSMs

santiregui, 2024-04-30:
Hi! We are migrating our robot control software (https://etasl.pages.gitlab.kuleuven.be/) from Orocos (developed at our lab) to ROS2 for a project and we are starting to use Yasmin to make the coordination between different robot skills and components. For our Orocos implementation we use rFSM (also developed at our lab): https://github.com/kmarkus/rFSM. 

So far so good :-) but I was wondering if there is a way to hide the visualization of nested FSMs from the yasmin_viewer. I think that being able to select which FSM to visualize is quite nice, but the parent/root/highest-level FSM always seems to depict the nested FSM, and this makes the visualization grow a lot and it becomes quite difficult to interpret. 

So, ideally for us, the visualization of the nested FSMs should be configurable. For our use case, if you are visualizing the parent FSM then the nested FSMs should be depicted as a single node (e.g. can be with a visual indicator that it is nested to avoid confusing it with a regular node). Then if you want to inspect the nested FSM you should be able to just select it in the visualizer. 

Is this already possible to configure or is it maybe planned to be developed? Otherwise a hint on which part of the visualizer's code to modify would be appreciated.

Thanks for your help!

mgonzs13, 2024-05-01:
Hi @santiregui, thanks for using YASMIN and for the suggestion. In fact, I have some pending updates for the viewer and I can include this. How would you add this feature, using a button to hide and show the content of all nested FSM? Btw, [here](https://github.com/uleroboticsgroup/yasmin/blob/main/yasmin_viewer/yasmin_viewer_web_client/src/components/FSM.js) you have the code that shows the FSM.

santiregui, 2024-05-01:
Hi @mgonzs13 , thanks a lot for your fast reply. That would be awesome!

Here are some ideas (brainstorming) in order from easiest to implement to cooler:

1) You could include a flag when you launch the visualizer in the terminal.
2) You could have one toggle button that activates the displaying of all nested FSMs in all FSM views.
3) You could have one toggle button per each FSM view that displays or hides all nested FSM associated to that particular FSM.
4) You could have a mechanism (e.g. toggle button, double clicking the node, "drop down" style icon, etc) at each nested node of all FSMs views that would allow you to display/expand or hide each individual nested FSM. That would make it more modular in the sense that you could have more control on what you want to visualize. Extra buttons of "expand all" and "hide all" may be needed to avoid cumbersome clickings.

These are just some ideas off the top of my head :-) maybe there is a better solution.

Cheers,
Santiago

mgonzs13, 2024-05-08:
Hi @santiregui, thanks for the suggestions. I'll take this into account when updating the viewer.

mgonzs13, 2024-05-27:
Hey @santiregui, here you have an example of the current state of the viewer. A checkbox is used to hide all nested FSMs.
[image]

santiregui, 2024-06-07:
Hi @mgonzs13 , thanks a lot for this implementation! I already tried it and it works like a charm :-).

 Maybe it would be useful to use a visual indicator that the state is a nested FSM when it is hidden (e.g. a different shape, other color of the contour of the elllipse, an icon, or whatever). But as it is it is really nice already, so thank you again! 

I also had to modify my code since I noticed that you got rid of the "simple node", but I like better the approach now which complies better with separation of concerns. Just as a recommendation, perhaps it is useful to log something in your examples, because I had to dig a bit into the code to discover the "YasminNode.get_instance()" mechanism. But is you put something trivial such as:

YasminNode.get_instance().get_logger().info("State A  called")

it might help people to get there faster (not only to log, but to do other stuff with the node).

Another idea (unrelated to the issue, sorry about that) that just came into my mind (just to brainstorm)  is to add a documentation field to the states, such that e.g. when you hover your mouse over a state you get some information about what the state is doing. An idea on how to implement this from the YASMIN user perspective:

sm.add_state("STATE_A", STATE_A(),
                    transitions={SUCCEED: "STATE_B"},
                    doc = "This handles the configuration of the robot...")

Cheers!

mgonzs13, 2024-06-07:
@santiregui, thanks for the ideas, I'll try to change the hidden nested state machines and I also want to try to fix the problem with the edges.

I removed simple_node since I want to have YASMIN independent of other packages. I also want to add more logs but I don't know what to do with the non-ROS 2 yasmin package.

Btw, I also like the idea of adding and showing state documentation. I have to check if cytoscape.js allows this easily. Another future work that I have noted is to show the blackboard of the state machines in the viewer.

AndersonYu7, 2024-07-29:
Hi! Seeing the nested state machine above, how do you do it?

Thanks！！

closed 2024-10-01 by mgonzs13