uleroboticsgroup/yasmin #18: Pause and resume an action state

WZHwerk5, 2024-05-14:
Hi, I just wonder how to pause/resume an action state with a ROS2 service. Thanks!

mgonzs13, 2024-05-14:
Hey @WZHwerk5, the action state involves using ROS 2 actions which, as far as I know, cannot be paused/resumed. However, pausing/resuming a state could be a good feature. Do you have any suggestions?

WZHwerk5, 2024-05-15:
Ok I see, thanks for quick reply! Let's step back a bit, how to cancel a running action state by a service (e.g. a Trigger)? Could you give me an example? Thanks in advance

mgonzs13, 2024-05-15:
In YASMIN, all states can be implemented to be canceled. In the case of the action states, canceling a state means canceling the action server. An example of action state is shown [here](https://github.com/uleroboticsgroup/yasmin/blob/main/yasmin_demo/yasmin_demo/action_client_demo.py). Thus, to cancel the FSM ```sm```, you can use the following code:

```python
sm.cancel_state()
```

mgonzs13, 2024-06-03:
Hey @WZHwerk5, how is this going?

WZHwerk5, 2024-06-03:
Hi, it worked, thanks.

jonyktan, 2024-10-10:
Hi, can I get some advice on how to cancel actions please? Given that the ActionState outcomes are (SUCCEED, ABORT, CANCEL), how can I trigger ABORT or CANCEL?

mgonzs13, 2024-10-10:
Hi @jonyktan, you have how to cancel an action in the previous comments.
> In YASMIN, all states can be implemented to be canceled. In the case of the action states, canceling a state means canceling the action server. An example of action state is shown [...]

> ```python
> sm.cancel_state()
> ```

jonyktan, 2024-10-10:
Could you give an example of where to insert `sm.cancel_state()` for an
event to trigger the state cancellation while in that state please? I'm not
too sure where to put it inside the ActionState

mgonzs13, 2024-10-10:
Let us consider the [action_client_demo.py](https://github.com/uleroboticsgroup/yasmin/blob/main/yasmin_demos/yasmin_demos/action_client_demo.py). In this example, a FSM is created with an ActionState. Then, the FSM is executed inside the main thread of the program.

To overview an example of canceling a FSM we can execute the FSM of the demo in a new thread as follows:
```python
    import threading
    t = threading.Thread(target=sm, args=(blackboard,))
    t.start()
    t.join()
```

This way, we can attempt to cancel the FSM while executing. Here you have the code modified canceling the FSM, for instance after 2 seconds of starting the execution:
```python
    import time
    import threading
    t = threading.Thread(target=sm, args=(blackboard,))
    t.start()
    time.sleep(2)
    sm.cancel_state()
    t.join()
```

I hope this is helpful to you. I'll add an example to the demo pkg.

jonyktan, 2025-01-03:
Hi again and happy new year! Sorry life happened and I only got back to this recently. 

I tested the latest [action_client_demo.py](https://github.com/uleroboticsgroup/yasmin/blob/main/yasmin_demos/yasmin_demos/action_client_demo.py) with the `sm.cancel_state()` implemented but noticed that the `KeyboardInterrupt` that triggers the `cancel_state()` kills the SM as well. 

Instead, I was hoping that the SM could accept a cancel state event while in ActionState that when triggered, would send a cancel_goal to the Fib action server and then transit back to its previous state, staying alive and ready for a new event to trigger the ActionState. 

So ideally:

```
    sm.add_state(
        "CALLING_FIBONACCI",
        FibonacciState(),
        transitions={
            SUCCEED: "PRINTING_RESULT",
            CANCEL: "<previous state>", # Event triggered cancellation of the action goal
            ABORT: "<another state>", # Action server triggered abort of the action
        },
    )
```

Do you have any suggestions on how I could do this in my state machine? Thanks for the help!

mgonzs13, 2025-01-06:
Hi @jonyktan, `KeyboardInterrupt` cancels the state machine since I want to stop all the demos. However, as mentioned before, you can create a service that cancels a state. What if you create a state, like the Fibonacci state, that inherits from `ActionState` and creates a service to cancel it?

jonyktan, 2025-01-08:
Hi @mgonzs13, thanks for your reply! I am not sure I understand what you mean - do you mean creating another ActionState (e.g. "CancelFibonacciState) within the same SM? While in FibonacciState, I wouldn't be able to transit to CancelFibonacciState until the Fibonacci action is complete though?

> In YASMIN, all states can be implemented to be canceled. In the case of the action states, canceling a state means canceling the action server. An example of action state is shown [...]

Also, I don't think cancelling the action should cancel the action server, as the action server should be independent of the action client (the `ActionState`).

jonyktan, 2025-01-08:
> Hi, it worked, thanks.

@WZHwerk5, would you be able to share a little on how you got it to work please? Thanks!

mgonzs13, 2025-01-08:
> Also, I don't think cancelling the action should cancel the action server, as the action server should be independent of the action client (the ActionState).

@jonyktan Cancling an action state will cancel the action client which means trying to cancel to action server.

One option is to cancel the state machine which will cancel the current state. You can use the following code inside a ROS 2 service:

```python3
sm.cancel_state()
```

Another option could be creating a service that cancels a specific state. For example, you can create this service inside the Fibonacci state as follow. However, take into account that having multiple instance of this state will create multiple service to cancel Fibonacci.

```python3
from yasmin_ros.yasmin_node import YasminNode
from std_srvs.srv import Trigger

class FibonacciState(ActionState):
    """
    Class representing the state of the Fibonacci action.

    Inherits from ActionState and implements methods to handle the
    Fibonacci action in a finite state machine.

    Attributes:
        None
    """

    def __init__(self) -> None:
        """
        Initializes the FibonacciState.

        Sets up the action type and the action name for the Fibonacci
        action. Initializes goal, response handler, and feedback
        processing callbacks.

        Parameters:
            None

        Returns:
            None
        """

        node = YasminNode.get_instance()
        self._cancel_srv = node.create_service(
            Trigger, "/cancel_fibonacci", self.cancel_fibonacci_cb
        )

        super().__init__(
            Fibonacci,  # action type
            "/fibonacci",  # action name
            self.create_goal_handler,  # callback to create the goal
            None,  # outcomes. Includes (SUCCEED, ABORT, CANCEL)
            self.response_handler,  # callback to process the response
            self.print_feedback,  # callback to process the feedback
        )

    def cancel_fibonacci_cb(
        self, req: Trigger.Request, res: Trigger.Response
    ) -> Trigger.Response:
        self.cancel_state()
        return res
```

closed 2025-02-22 by github-actions[bot]