uleroboticsgroup/yasmin #19: Not able to run the demo

amodpatil1, 2024-06-04:
colcon build --packages-select yasmin
ros2 run yasmin_demo yasmin_demo.py
Package 'yasmin_demo' not found
not able to run the command. please help me out.

mgonzs13, 2024-06-04:
Hey @amodpatil1, have you source (```source install/setup.bash```) after using colcon?

amodpatil1, 2024-06-05:
Yes, I did source the workspace before the running the command and I tried it multiple times and I also tried doing from start as well, but it did not work.

mgonzs13, 2024-06-05:
Have you colcon all the yasmin packages (yasmin, yasmin_msgs, yasmin_demo, yasmin_ros, yasmin_viewe) or only yasmin?

amodpatil1, 2024-06-05:
I did it for yasmin only. Shouldn't the entire package get built when I do Colcon build?

mgonzs13, 2024-06-05:
No, since they are different packages.

amodpatil1, 2024-06-05:
okay I will try that.

amodpatil1, 2024-06-05:
Starting >>> yasmin_demo
--- stderr: yasmin_demo                             
In file included from /home/amodpatil/ros2/src/yasmin/yasmin_demo/src/action_client_demo.cpp:24:
/home/amodpatil/ros2/install/yasmin_ros/include/yasmin_ros/action_state.hpp: In instantiation of ‘std::string yasmin_ros::ActionState<ActionT>::execute(std::shared_ptr<yasmin::blackboard::Blackboard>) [with ActionT = action_tutorials_interfaces::action::Fibonacci; std::string = std::__cxx11::basic_string<char>]’:
/home/amodpatil/ros2/install/yasmin_ros/include/yasmin_ros/action_state.hpp:103:3:   required from here
/home/amodpatil/ros2/install/yasmin_ros/include/yasmin_ros/action_state.hpp:113:46: error: no match for ‘operator=’ (operand types are ‘rclcpp_action::Client<action_tutorials_interfaces::action::Fibonacci>::GoalResponseCallback’ {aka ‘std::function<void(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<action_tutorials_interfaces::action::Fibonacci> > >)>’} and ‘std::_Bind_helper<false, void (yasmin_ros::ActionState<action_tutorials_interfaces::action::Fibonacci>::*)(const std::shared_ptr<rclcpp_action::ClientGoalHandle<action_tutorials_interfaces::action::Fibonacci> >&), yasmin_ros::ActionState<action_tutorials_interfaces::action::Fibonacci>*, const std::_Placeholder<1>&>::type’ {aka ‘std::_Bind<void (yasmin_ros::ActionState<action_tutorials_interfaces::action::Fibonacci>::*(yasmin_ros::ActionState<action_tutorials_interfaces::action::Fibonacci>*, std::_Placeholder<1>))(const std::shared_ptr<rclcpp_action::ClientGoalHandle<action_tutorials_interfaces::action::Fibonacci> >&)>’})
  113 |     send_goal_options.goal_response_callback =
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
  114 |         std::bind(&ActionState::goal_response_callback, this, _1);
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/c++/9/future:48,
                 from /opt/ros/foxy/include/rclcpp/executors.hpp:18,
                 from /opt/ros/foxy/include/rclcpp/rclcpp.hpp:146,
                 from /home/amodpatil/ros2/install/yasmin_ros/include/yasmin_ros/action_state.hpp:23,
                 from /home/amodpatil/ros2/src/yasmin/yasmin_demo/src/action_client_demo.cpp:24:
/usr/include/c++/9/bits/std_function.h:462:7: note: candidate: ‘std::function<_Res(_ArgTypes ...)>& std::function<_Res(_ArgTypes ...)>::operator=(const std::function<_Res(_ArgTypes ...)>&) [with _Res = void; _ArgTypes = {std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<action_tutorials_interfaces::action::Fibonacci> > >}]’
  462 |       operator=(const function& __x)
      |       ^~~~~~~~
/usr/include/c++/9/bits/std_function.h:462:33: note:   no known conversion for argument 1 from ‘std::_Bind_helper<false, void (yasmin_ros::ActionState<action_tutorials_interfaces::action::Fibonacci>::*)(const std::shared_ptr<rclcpp_action::ClientGoalHandle<action_tutorials_interfaces::action::Fibonacci> >&), yasmin_ros::ActionState<action_tutorials_interfaces::action::Fibonacci>*, const std::_Placeholder<1>&>::type’ {aka ‘std::_Bind<void (yasmin_ros::ActionState<action_tutorials_interfaces::action::Fibonacci>::*(yasmin_ros::ActionState<action_tutorials_interfaces::action::Fibonacci>*, std::_Placeholder<1>))(const std::shared_ptr<rclcpp_action::ClientGoalHandle<action_tutorials_interfaces::action::Fibonacci> >&)>’} to ‘const std::function<void(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<action_tutorials_interfaces::action::Fibonacci> > >)>&’
  462 |       operator=(const function& __x)
      |                 ~~~~~~~~~~~~~~~~^~~
/usr/include/c++/9/bits/std_function.h:480:7: note: candidate: ‘std::function<_Res(_ArgTypes ...)>& std::function<_Res(_ArgTypes ...)>::operator=(std::function<_Res(_ArgTypes ...)>&&) [with _Res = void; _ArgTypes = {std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<action_tutorials_interfaces::action::Fibonacci> > >}]’
  480 |       operator=(function&& __x) noexcept
      |       ^~~~~~~~
/usr/include/c++/9/bits/std_function.h:480:28: note:   no known conversion for argument 1 from ‘std::_Bind_helper<false, void (yasmin_ros::ActionState<action_tutorials_interfaces::action::Fibonacci>::*)(const std::shared_ptr<rclcpp_action::ClientGoalHandle<action_tutorials_interfaces::action::Fibonacci> >&), yasmin_ros::ActionState<action_tutorials_interfaces::action::Fibonacci>*, const std::_Placeholder<1>&>::type’ {aka ‘std::_Bind<void (yasmin_ros::ActionState<action_tutorials_interfaces::action::Fibonacci>::*(yasmin_ros::ActionState<action_tutorials_interfaces::action::Fibonacci>*, std::_Placeholder<1>))(const std::shared_ptr<rclcpp_action::ClientGoalHandle<action_tutorials_interfaces::action::Fibonacci> >&)>’} to ‘std::function<void(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<action_tutorials_interfaces::action::Fibonacci> > >)>&&’
  480 |       operator=(function&& __x) noexcept
      |                 ~~~~~~~~~~~^~~
/usr/include/c++/9/bits/std_function.h:494:7: note: candidate: ‘std::function<_Res(_ArgTypes ...)>& std::function<_Res(_ArgTypes ...)>::operator=(std::nullptr_t) [with _Res = void; _ArgTypes = {std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<action_tutorials_interfaces::action::Fibonacci> > >}; std::nullptr_t = std::nullptr_t]’
  494 |       operator=(nullptr_t) noexcept
      |       ^~~~~~~~
[... cut ...]
make[2]: *** [CMakeFiles/action_client_demo.dir/build.make:63: CMakeFiles/action_client_demo.dir/src/action_client_demo.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:84: CMakeFiles/action_client_demo.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:141: all] Error 2
---
Failed   <<< yasmin_demo [7.38s, exited with code 2]

Summary: 0 packages finished [7.67s]
  1 package failed: yasmin_demo
  1 package had stderr output: yasmin_demo
  
  I built all other packages but the demo is only facing an issue

mgonzs13, 2024-06-05:
That is a problem with the ROS 2 distro. Foxy EOL was on June 20th, 2023. Thus, yasmin was updated to Humble.

mgonzs13, 2024-06-05:
I have just added support for Foxy. It should work now.

amodpatil1, 2024-06-05:
okay I will try

amodpatil1, 2024-06-05:
--- stderr: yasmin_demo                             
In file included from /home/amodpatil/ros2/src/yasmin/yasmin_demo/src/action_client_demo.cpp:24:
/home/amodpatil/ros2/install/yasmin_ros/include/yasmin_ros/action_state.hpp: In instantiation of ‘std::string yasmin_ros::ActionState<ActionT>::execute(std::shared_ptr<yasmin::blackboard::Blackboard>) [with ActionT = action_tutorials_interfaces::action::Fibonacci; std::string = std::__cxx11::basic_string<char>]’:
/home/amodpatil/ros2/install/yasmin_ros/include/yasmin_ros/action_state.hpp:103:3:   required from here
/home/amodpatil/ros2/install/yasmin_ros/include/yasmin_ros/action_state.hpp:113:46: error: no match for ‘operator=’ (operand types are ‘rclcpp_action::Client<action_tutorials_interfaces::action::Fibonacci>::GoalResponseCallback’ {aka ‘std::function<void(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<action_tutorials_interfaces::action::Fibonacci> > >)>’} and ‘std::_Bind_helper<false, void (yasmin_ros::ActionState<action_tutorials_interfaces::action::Fibonacci>::*)(const std::shared_ptr<rclcpp_action::ClientGoalHandle<action_tutorials_interfaces::action::Fibonacci> >&), yasmin_ros::ActionState<action_tutorials_interfaces::action::Fibonacci>*, const std::_Placeholder<1>&>::type’ {aka ‘std::_Bind<void (yasmin_ros::ActionState<action_tutorials_interfaces::action::Fibonacci>::*(yasmin_ros::ActionState<action_tutorials_interfaces::action::Fibonacci>*, std::_Placeholder<1>))(const std::shared_ptr<rclcpp_action::ClientGoalHandle<action_tutorials_interfaces::action::Fibonacci> >&)>’})
  113 |     send_goal_options.goal_response_callback =
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
  114 |         std::bind(&ActionState::goal_response_callback, this, _1);
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/c++/9/future:48,
                 from /opt/ros/foxy/include/rclcpp/executors.hpp:18,
                 from /opt/ros/foxy/include/rclcpp/rclcpp.hpp:146,
                 from /home/amodpatil/ros2/install/yasmin_ros/include/yasmin_ros/action_state.hpp:23,
                 from /home/amodpatil/ros2/src/yasmin/yasmin_demo/src/action_client_demo.cpp:24:
/usr/include/c++/9/bits/std_function.h:462:7: note: candidate: ‘std::function<_Res(_ArgTypes ...)>& std::function<_Res(_ArgTypes ...)>::operator=(const std::function<_Res(_ArgTypes ...)>&) [with _Res = void; _ArgTypes = {std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<action_tutorials_interfaces::action::Fibonacci> > >}]’
  462 |       operator=(const function& __x)
      |       ^~~~~~~~
/usr/include/c++/9/bits/std_function.h:462:33: note:   no known conversion for argument 1 from ‘std::_Bind_helper<false, void (yasmin_ros::ActionState<action_tutorials_interfaces::action::Fibonacci>::*)(const std::shared_ptr<rclcpp_action::ClientGoalHandle<action_tutorials_interfaces::action::Fibonacci> >&), yasmin_ros::ActionState<action_tutorials_interfaces::action::Fibonacci>*, const std::_Placeholder<1>&>::type’ {aka ‘std::_Bind<void (yasmin_ros::ActionState<action_tutorials_interfaces::action::Fibonacci>::*(yasmin_ros::ActionState<action_tutorials_interfaces::action::Fibonacci>*, std::_Placeholder<1>))(const std::shared_ptr<rclcpp_action::ClientGoalHandle<action_tutorials_interfaces::action::Fibonacci> >&)>’} to ‘const std::function<void(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<action_tutorials_interfaces::action::Fibonacci> > >)>&’
  462 |       operator=(const function& __x)
      |                 ~~~~~~~~~~~~~~~~^~~
/usr/include/c++/9/bits/std_function.h:480:7: note: candidate: ‘std::function<_Res(_ArgTypes ...)>& std::function<_Res(_ArgTypes ...)>::operator=(std::function<_Res(_ArgTypes ...)>&&) [with _Res = void; _ArgTypes = {std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<action_tutorials_interfaces::action::Fibonacci> > >}]’
  480 |       operator=(function&& __x) noexcept
      |       ^~~~~~~~
/usr/include/c++/9/bits/std_function.h:480:28: note:   no known conversion for argument 1 from ‘std::_Bind_helper<false, void (yasmin_ros::ActionState<action_tutorials_interfaces::action::Fibonacci>::*)(const std::shared_ptr<rclcpp_action::ClientGoalHandle<action_tutorials_interfaces::action::Fibonacci> >&), yasmin_ros::ActionState<action_tutorials_interfaces::action::Fibonacci>*, const std::_Placeholder<1>&>::type’ {aka ‘std::_Bind<void (yasmin_ros::ActionState<action_tutorials_interfaces::action::Fibonacci>::*(yasmin_ros::ActionState<action_tutorials_interfaces::action::Fibonacci>*, std::_Placeholder<1>))(const std::shared_ptr<rclcpp_action::ClientGoalHandle<action_tutorials_interfaces::action::Fibonacci> >&)>’} to ‘std::function<void(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<action_tutorials_interfaces::action::Fibonacci> > >)>&&’
  480 |       operator=(function&& __x) noexcept
      |                 ~~~~~~~~~~~^~~
/usr/include/c++/9/bits/std_function.h:494:7: note: candidate: ‘std::function<_Res(_ArgTypes ...)>& std::function<_Res(_ArgTypes ...)>::operator=(std::nullptr_t) [with _Res = void; _ArgTypes = {std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<action_tutorials_interfaces::action::Fibonacci> > >}; std::nullptr_t = std::nullptr_t]’
  494 |       operator=(nullptr_t) noexcept
      |       ^~~~~~~~
[... cut ...]
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/action_client_demo.dir/build.make:63: CMakeFiles/action_client_demo.dir/src/action_client_demo.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:84: CMakeFiles/action_client_demo.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:141: all] Error 2
---
Failed   <<< yasmin_demo [6.82s, exited with code 2]

Summary: 0 packages finished [7.10s]
  1 package failed: yasmin_demo
  1 package had stderr output: yasmin_demo

still the same error

mgonzs13, 2024-06-05:
Pull and try again. I have done another update.

amodpatil1, 2024-06-05:
It is working now. But now there is this ros2 run demo_nodes_py add_two_ints_servers
No executable found. I am sorry for too many questions

amodpatil1, 2024-06-05:
The service demo is not running

amodpatil1, 2024-06-05:
Both the commands are not executable

mgonzs13, 2024-06-05:
You have to install the ROS 2 demo packages (ros-humble-demo-nodes-py, ros-humble-demo-nodes-cpp, ros-humble-action-tutorials-py, ros-humble-action-tutorials-cpp).

amodpatil1, 2024-06-05:
I am able to find the demos for foxy, can you please help me out?

amodpatil1, 2024-06-05:
Yes it started working but I am not able to understand how can I build a custom FSM on yasmin. Could we please have a call on this where you can explain me the implementation of yasmin on my project?
I am a Masters student at Coburg University of applied science in germany and I am currently working on an autonomous vehicle, where the vehicle needs to park itself autonomously in the parking spot and I am working on the behaviour planning component which decides the state of the vehicle.

mgonzs13, 2024-06-06:
@amodpatil1, you have several [examples](https://github.com/uleroboticsgroup/yasmin?tab=readme-ov-file#demos) in the README that includes basic states, ROS 2-based states and a demo with Nav2. If you want further explanations, you can send me an email.

amodpatil1, 2024-06-06:
Do I need to have a separate package for my state machine?

mgonzs13, 2024-06-06:
Yes, you should create a new package with your code.

amodpatil1, 2024-06-06:
do I need to add the dependencies from yasmin into my setup.py as well?

mgonzs13, 2024-06-06:
No, but you can add them into the package.xml.

amodpatil1, 2024-06-07:
# Copyright (C) 2023  Miguel Ángel González Santamarta

# This program is free software: you can redistribute it and/or modify
# it under the terms of the GNU General Public License as published by
# the Free Software Foundation, either version 3 of the License, or
# (at your option) any later version.

# This program is distributed in the hope that it will be useful,
# but WITHOUT ANY WARRANTY; without even the implied warranty of
# MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
# GNU General Public License for more details.

# You should have received a copy of the GNU General Public License
# along with this program.  If not, see <https://www.gnu.org/licenses/>.

from typing import Dict, List, Union
from threading import Lock
from yasmin.state import State
from yasmin.blackboard import Blackboard

class StateMachine(State):
    def __init__(self, outcomes: List[str]) -> None:

        super().__init__(outcomes)

        self._states = {}
        self._start_state = None
        self.__current_state = None
        self.__current_state_lock = Lock()

    def add_state(
        self,
        name: str,
        state: State,
        transitions: Dict[str, str] = None
    ) -> None:

        if not transitions:
            transitions = {}

        self._states[name] = {
            "state": state,
            "transitions": transitions
        }

        if not self._start_state:
            self._start_state = name

    def set_start_state(self, name: str) -> None:
        self._start_state = name

    def get_start_state(self) -> str:
        return self._start_state

    def cancel_state(self) -> None:
        super().cancel_state()
        with self.__current_state_lock:
            if self.__current_state:
                self._states[self.__current_state]["state"].cancel_state()

    def execute(self, blackboard: Blackboard) -> str:

        with self.__current_state_lock:
            self.__current_state = self._start_state

        while True:

            with self.__current_state_lock:
                state = self._states[self.__current_state]

            outcome = state["state"](blackboard)

            # check outcome belongs to state
            if outcome not in state["state"].get_outcomes():
                raise Exception(
                    f"Outcome ({outcome}) is not register in state {self.__current_state}")

            # translate outcome using transitions
            if outcome in state["transitions"]:
                outcome = state["transitions"][outcome]

            # outcome is an outcome of the sm
            if outcome in self.get_outcomes():
                with self.__current_state_lock:
                    self.__current_state = None
                return outcome

            # outcome is a state
            elif outcome in self._states:
                with self.__current_state_lock:
                    self.__current_state = outcome

            # outcome is not in the sm
            else:
                raise Exception(f"Outcome ({outcome}) without transition")

    def get_states(self) -> Dict[str, Union[State, Dict[str, str]]]:
        return self._states

    def get_current_state(self) -> str:
        with self.__current_state_lock:
            if self.__current_state:
                return self.__current_state

        return ""

    def __str__(self) -> str:
        return f"StateMachine: {self._states}"
        
        
        Can I use this as reference for my custom state machnine?

amodpatil1, 2024-06-07:
Can I use this as a reference for my custom machine?

# Copyright (C) 2023  Miguel Ángel González Santamarta

# This program is free software: you can redistribute it and/or modify
# it under the terms of the GNU General Public License as published by
# the Free Software Foundation, either version 3 of the License, or
# (at your option) any later version.

# This program is distributed in the hope that it will be useful,
# but WITHOUT ANY WARRANTY; without even the implied warranty of
# MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
# GNU General Public License for more details.

# You should have received a copy of the GNU General Public License
# along with this program.  If not, see <https://www.gnu.org/licenses/>.

from typing import Dict, List, Union
from threading import Lock
from yasmin.state import State
from yasmin.blackboard import Blackboard

class StateMachine(State):
    def __init__(self, outcomes: List[str]) -> None:

        super().__init__(outcomes)

        self._states = {}
        self._start_state = None
        self.__current_state = None
        self.__current_state_lock = Lock()

    def add_state(
        self,
        name: str,
        state: State,
        transitions: Dict[str, str] = None
    ) -> None:

        if not transitions:
            transitions = {}

        self._states[name] = {
            "state": state,
            "transitions": transitions
        }

        if not self._start_state:
            self._start_state = name

    def set_start_state(self, name: str) -> None:
        self._start_state = name

    def get_start_state(self) -> str:
        return self._start_state

    def cancel_state(self) -> None:
        super().cancel_state()
        with self.__current_state_lock:
            if self.__current_state:
                self._states[self.__current_state]["state"].cancel_state()

    def execute(self, blackboard: Blackboard) -> str:

        with self.__current_state_lock:
            self.__current_state = self._start_state

        while True:

            with self.__current_state_lock:
                state = self._states[self.__current_state]

            outcome = state["state"](blackboard)

            # check outcome belongs to state
            if outcome not in state["state"].get_outcomes():
                raise Exception(
                    f"Outcome ({outcome}) is not register in state {self.__current_state}")

            # translate outcome using transitions
            if outcome in state["transitions"]:
                outcome = state["transitions"][outcome]

            # outcome is an outcome of the sm
            if outcome in self.get_outcomes():
                with self.__current_state_lock:
                    self.__current_state = None
                return outcome

            # outcome is a state
            elif outcome in self._states:
                with self.__current_state_lock:
                    self.__current_state = outcome

            # outcome is not in the sm
            else:
                raise Exception(f"Outcome ({outcome}) without transition")

    def get_states(self) -> Dict[str, Union[State, Dict[str, str]]]:
        return self._states

    def get_current_state(self) -> str:
        with self.__current_state_lock:
            if self.__current_state:
                return self.__current_state

        return ""

    def __str__(self) -> str:
        return f"StateMachine: {self._states}"

mgonzs13, 2024-06-07:
> What would be the effieicent way to implement yasmin? Would it be via creating a package or a YAML file?

I prefer using new packages. What do you mean by a YAML file?

> Can I use this as a reference for my custom machine?

What do you mean by reference?

amodpatil1, 2024-06-08:
I mean can I write my state machine using the one you wrote as a base for mine custom state machine

amodpatil1, 2024-06-08:
states:
  - id: idle
    on_enter:
      - log: Entered idle state
      - custom_action: adapt_vi_start
      - custom_action: adapt_loc_start
      - custom_action: adapt_obj_start
      - custom_action: adapt_envmod_start
      - custom_action: ylidar_start
      - custom_action: adapt_livtrac_start
    transitions:
      - trigger: start
        target: driving

  - id: driving
    on_enter:
      - log: Entered driving state
      - custom_action: adapt_vi_drive
      - custom_action: adapt_loc_drive
      - custom_action: adapt_obj_drive
      - custom_action: adapt_envmod_drive
      - custom_action: ylidar_drive
      - custom_action: adapt_livtrac_drive
    transitions:
      - trigger: stop
        target: idle

This is the example for the yaml file
I have not implemented this

amodpatil1, 2024-06-08:
I'm really having implementing this to a custom machine

mgonzs13, 2024-06-09:
You can use any of the state machines from the demos but I recommend you create a new ROS 2 package with your code.

amodpatil1, 2024-06-12:
Okay I will try it

mgonzs13, 2024-06-12:
The blackboard is shared between all states and nested state machines. You can create a blackboard object if you want to provide some initial data for your main state machine. Otherwise, you can avoid creating it and you can use the default blackboard in your states.

amodpatil1, 2024-06-12:
I created the code for my custom state machine but now the issue is that I am getting the following error "Traceback (most recent call last):
  File "/home/amodpatil/ros2/src/behstate/behstate/behmac.py", line 21, in <module>
    from yasmin import State
ModuleNotFoundError: No module named 'yasmin'"

mgonzs13, 2024-06-12:
That seems a problem with colcon and source. How are you running your state machine, using ```ros2 run```?

amodpatil1, 2024-06-12:
python3 behmac.py

mgonzs13, 2024-06-12:
That should also work. Make sure to colcon all yasmin package and use ```source install/setup.bash```.

amodpatil1, 2024-06-12:
It ran thank you so much

amodpatil1, 2024-06-12:
if I have any doubts, how can I reach you?

mgonzs13, 2024-06-13:
Please feel free to write me here or to my email (the one shown in my account).

amodpatil1, 2024-06-13:
Okay, I have one doubt if I want to use multiple components within the state machine then how can I incorporate them within my code?
#!/usr/bin/env python3

# Copyright (C) 2023  Miguel Ángel González Santamarta

# This program is free software: you can redistribute it and/or modify
# it under the terms of the GNU General Public License as published by
# the Free Software Foundation, either version 3 of the License, or
# (at your option) any later version.

# This program is distributed in the hope that it will be useful,
# but WITHOUT ANY WARRANTY; without even the implied warranty of
# MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
# GNU General Public License for more details.

# You should have received a copy of the GNU General Public License
# along with this program.  If not, see <https://www.gnu.org/licenses/>.

import time
import math
import rclpy
from yasmin import State
from yasmin import Blackboard
from yasmin import StateMachine
from yasmin_viewer import YasminViewerPub

# define state Idle
class IdleState(State):
    def __init__(self) -> None:
        super().__init__(["driving_state"])

    def execute(self, blackboard: Blackboard) -> str:
        print("Executing state IDLE")
        # Simulate waiting for start trigger from adapt_vi
        while not blackboard.adapt_vi_start_trigger:
            time.sleep(0.1)
        return "driving_state"

# define state Driving
class DrivingState(State):
    def __init__(self) -> None:
        super().__init__(["drive", "stop_obstacle", "stop_near_park"])

    def execute(self, blackboard: Blackboard) -> str:
        print("Executing state DRIVING")
        time.sleep(3)

        # Dummy condition checks
        if blackboard.adapt_envmod:
            print("Obstacle detected by adapt_envmod")
            return "stop_obstacle"

        # Calculate distance to parking spot
        vehicle_location = blackboard.adapt_loc
        parking_spot_location = blackboard.adapt_roucomp
        distance_to_parking_spot = math.sqrt(
            (parking_spot_location[0] - vehicle_location[0]) ** 2 +
            (parking_spot_location[1] - vehicle_location[1]) ** 2
        )

        if distance_to_parking_spot < blackboard.parking_threshold:
            print("Near parking spot")
            return "stop_near_park"

        return "drive"

# define state StopObstacle
class StopObstacleState(State):
    def __init__(self) -> None:
        super().__init__(["idle"])

    def execute(self, blackboard: Blackboard) -> str:
        print("Executing state STOP_OBSTACLE")
        time.sleep(2)
        return "idle"

# define state StopNearPark
class StopNearParkState(State):
    def __init__(self) -> None:
        super().__init__(["idle"])

    def execute(self, blackboard: Blackboard) -> str:
        print("Executing state STOP_NEAR_PARK")
        time.sleep(2)
        return "idle"

# main
def main():

    print("yasmin_demo")

    # init ROS 2
    rclpy.init()

    # create a FSM
    sm = StateMachine(outcomes=["finished"])

    # create and set blackboard
    blackboard = Blackboard()
    blackboard.obstacle_detected = False
    blackboard.near_parking_spot = False
    blackboard.adapt_vi_start_trigger = False  # Initial state of the start trigger
    blackboard.adapt_loc = (0, 0)  # Initialize adapt_loc with a tuple representing coordinates
    blackboard.adapt_obj = None  # Initialize adapt_obj
    blackboard.adapt_livtrac = None  # Initialize adapt_livtrac
    blackboard.adapt_roucomp = (10, 10)  # Initialize adapt_roucomp with a dummy parking spot location
    blackboard.adapt_trajp = None  # Initialize adapt_trajp
    blackboard.adapt_envmod = False  # Initialize adapt_envmod as no obstacle detected
    blackboard.parking_threshold = 0.5  # Distance threshold for parking spot detection

    # add states
    sm.add_state(
        "IDLE",
        IdleState(),
        transitions={
            "driving_state": "DRIVING"
        }
    )
    sm.add_state(
        "DRIVING",
        DrivingState(),
        transitions={
            "drive": "DRIVING",
            "stop_obstacle": "STOP_OBSTACLE",
            "stop_near_park": "STOP_NEAR_PARK"
        }
    )
    sm.add_state(
        "STOP_OBSTACLE",
        StopObstacleState(),
        transitions={
            "idle": "IDLE"
        }
    )
    sm.add_state(
        "STOP_NEAR_PARK",
        StopNearParkState(),
        transitions={
            "idle": "IDLE"
        }
    )

    # pub FSM info
    YasminViewerPub("YASMIN_DEMO", sm)

    # Set the start trigger to True after a delay to simulate an external trigger
    time.sleep(2)
    blackboard.adapt_vi_start_trigger = True

    # Simulate vehicle driving and detecting an obstacle and nearing parking spot
    time.sleep(5)
    blackboard.adapt_envmod = True  # Simulate obstacle detection

    # execute FSM
    outcome = sm.execute(blackboard)
    print(outcome)

    # shutdown ROS 2
    rclpy.shutdown()

if __name__ == "__main__":
    main()

mgonzs13, 2024-06-13:
What do you mean by multiple components?

amodpatil1, 2024-06-13:
[Autonomous Driving And Parking Technology - Git HS-Coburg.pdf](https://github.com/user-attachments/files/15820175/Autonomous.Driving.And.Parking.Technology.-.Git.HS-Coburg.pdf)
This is my project repositary and in this there are multiple components

mgonzs13, 2024-06-13:
I am not getting what do you mean by components. If you mean other ROS 2 nodes, you can use YASMIN states. If you want to use other topics, actions or services, you can use the ROS 2-based states (ServiceState, ActionState). For instance, you can have a state for the navigation/driving. [Here](https://github.com/uleroboticsgroup/yasmin/blob/main/yasmin_demo/yasmin_demo/nav_demo.py) you have an example using Navigation2. Besides, you can subscribe, such as subscribing to the object detection topic; or publish, for example, twist msgs to control you car.

amodpatil1, 2024-06-13:
Yes I mean multiple nodes and I want to subscribe to multiple nodes at a time which are mentioned in the Repository. I am making a change in the https://github.com/uleroboticsgroup/yasmin/blob/main/yasmin_ros/yasmin_ros/yasmin_node.py in order to subscribe to multiple topics simultaneously.

mgonzs13, 2024-06-13:
Why do you need to modify it? You can use it as a ROS 2 node to create multiple subscribers or you can use the MonitorState.

amodpatil1, 2024-06-13:
me and professor discussed on this and he sent me a code for this https://github.com/uleroboticsgroup/yasmin/blob/main/yasmin_ros/yasmin_ros/yasmin_node.py since I was not able to subscribe to a topic and which was supposed change the state
from Idle to driving 
https://github.com/uleroboticsgroup/yasmin/issues/19#issuecomment-2165073540

mgonzs13, 2024-06-14:
You don't need to modify or create new nodes. You have two options to subscribe to a topic using YASMIN:

 - Using [MonitorState](https://github.com/uleroboticsgroup/yasmin/blob/main/yasmin_ros/yasmin_ros/monitor_state.py) which subscribes to a topic and allows you to treat messages and return different outcomes. [Here](https://github.com/uleroboticsgroup/yasmin/blob/main/yasmin_demo/yasmin_demo/monitor_demo.py) you have an example with odom msgs.
 - Creating a state that uses the yasmin_node similar to the ROS 2-based states. 
 
 ```python
from yasmin import State
from yasmin_ros.yasmin_node import YasminNode

class TopicState(State):
  def __init__(self):
    super().__init__(outcomes)
    self._node = YasminNode.get_instance()
    self._sub = self._node.create_subscription(
              msg_type, topic_name, self.__callback, qos)
```

amodpatil1, 2024-06-14:
Yes I mean multiple components and I want to use topics

amodpatil1, 2024-06-14:
I am facing this error

Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[WARN]

mgonzs13, 2024-06-14:
As stated in the log, you are using two nodes with the same name. Which nodes are you using?

amodpatil1, 2024-06-17:
I have changed the code now and still I am not able listen to the topic

mgonzs13, 2024-06-17:
So now what are you using, the MonitorState or a new node? Have you checked if there are msgs using echo? Are you using the correct QoS?

amodpatil1, 2024-06-17:
I am using a new node and I checked using ros2 topic echo but the node is not able to listen to that topic

mgonzs13, 2024-06-17:
Unfortunately, I can't help you without taking a look at your code. You can send it to me by email. The QoS of the topics may also affect if the subscriber and the publisher has different QoS.

amodpatil1, 2024-06-17:
I have sent you the code via email. The email id from which I have sent you the email is amodpatil90@gmail.com

amodpatil1, 2024-06-17:
Are you able to listen to multiple topics simultaneously?

amodpatil1, 2024-06-17:
Can you send me the screenshot of it?

mgonzs13, 2024-06-17:
Here you have a video. I am pushing msgs in different topics and different tabs. Is this what you want?

https://github.com/uleroboticsgroup/yasmin/assets/25979134/2c39246a-5acf-4d76-8037-ad36108cf0b6

amodpatil1, 2024-06-17:
Yes this is what I want. Did you do any changes in the yasmin_node?

mgonzs13, 2024-06-17:
No, I haven't changed anything. I have just executed it as you gave me.

[24 more replies]

closed 2024-08-17 by amodpatil1