uleroboticsgroup/yasmin #20: Change state from one to another with a service call?

Devansh1902, 2024-06-06:
how can user change state from foo to bar using a service call 
please help

mgonzs13, 2024-06-06:
Hi @Devansh1902, you can create an idle state that waits for service requests and change to another state when a request is received. Is this what you want?

Devansh1902, 2024-06-06:
Yes lets say foo state is the start state and from foo i want to go to bar with a service call and when in bar state also the transition should happen from another service call to change the state back to foo

mgonzs13, 2024-06-06:
So, in that case, should both states have a service or do you want only one service? For the first case, a new ROS 2-based state can be created, similar to the action and service client states.

Devansh1902, 2024-06-06:
Only one service in the default state and from that default state user can do a service call to change the state and then stay in the state till another service call comes to go back to default state

mgonzs13, 2024-06-06:
You can create a node with a service that controls the switch between the states by using a variable on the blackboard. So the states should wait for the service call.

Devansh1902, 2024-06-06:
please explain this line of code 
  std::string outcome = (*sm.get())();

mgonzs13, 2024-06-06:
StateMachine object is got from the pointer and its operator () is called which returns an outcome.

Devansh1902, 2024-06-06:
i have several states but one is default one :

auto sm = std::make_shared<yasmin::StateMachine>(yasmin::StateMachine({""}));
    // add states
    sm->add_state("Standby", std::make_shared<StandbyState>(), {{"outcome0", "Standby"}, {"outcome1", "manual"}, {"outcome2",      "   teleop"}, {"outcome3", "assisted"}, {"outcome4", "convoy"}, {"outcome5", "waypoint"}});
    sm->add_state("manual", std::make_shared<ManualModeState>(), {{"outcome6", "Standby"}});
    sm->add_state("teleop", std::make_shared<TeleopState>(), {{"outcome10", "Standby"}});
    sm->add_state("assisted", std::make_shared<AssistedTeleopState>(), {{"outcome7", "Standby"}});
    sm->add_state("convoy", std::make_shared<ConvoyFollowState>(), {{"outcome8", "Standby"}});
    sm->add_state("waypoint", std::make_shared<WaypointNavigationState>(), {{"outcome9", "Standby"}});
     // execute
    std::string outcome = (*sm.get())();

mgonzs13, 2024-06-06:
You should add an outcome to your state machine (```auto sm = std::make_sharedyasmin::StateMachine(yasmin::StateMachine({"End"}));```). Then, you can add a transition from your Standby state to that outcome (```{"outcome6", "End"}```).

Devansh1902, 2024-06-06:
sorry i didnt got you. for what we ll use the end ?

mgonzs13, 2024-06-06:
You have to add the final outcomes of the state machine, similar to the states, for instance, "End". Check the [demo](https://github.com/uleroboticsgroup/yasmin/blob/main/yasmin_demo/src/yasmin_demo.cpp).

closed 2024-12-29 by github-actions[bot]