uleroboticsgroup/yasmin #25: Error during build: src/yasmin/state.cpp: error: ‘logic_error’ is not a member of ‘std’

mowoe, 2024-10-09:
Hi,

thanks for your efforts with this library! I am going to preface this with saying that I am relatively new to C++ (and colcon even more), so I apologize if this is an obvious oversight on my part. I followed the #Install section in your readme and tried to compile my workspace with `colcon build`.

While e.g. `yasmin_msgs` builds fine, the main `yasmin` package fails with numerous errors like this:
```bash
/workspaces/mcu-ros/src/yasmin/yasmin/include/yasmin/blackboard/blackboard.hpp:25:1: note: ‘std::runtime_error’ is defined in header ‘<stdexcept>’; did you forget to ‘#include <stdexcept>’?
   24 | #include "yasmin/blackboard/blackboard_value.hpp"
  +++ |+#include <stdexcept>
   25 | #include "yasmin/blackboard/blackboard_value_interface.hpp"
/workspaces/mcu-ros/src/yasmin/yasmin/src/yasmin/state.cpp: In member function ‘std::string yasmin::State::operator()(std::shared_ptr<yasmin::blackboard::Blackboard>)’:
/workspaces/mcu-ros/src/yasmin/yasmin/src/yasmin/state.cpp:34:16: error: ‘logic_error’ is not a member of ‘std’
   34 |     throw std::logic_error("Outcome (" + outcome + ") does not exist");
      |                ^~~~~~~~~~~
/workspaces/mcu-ros/src/yasmin/yasmin/src/yasmin/state.cpp:22:1: note: ‘std::logic_error’ is defined in header ‘<stdexcept>’; did you forget to ‘#include <stdexcept>’?
   21 | #include "yasmin/state.hpp"
  +++ |+#include <stdexcept>
   22 | 
```

I already tried to re-source the environment, which gives me this error:
```bash
$ . install/setup.bash 
not found: "/workspaces/mcu-ros/install/yasmin/share/yasmin/local_setup.bash"
not found: "/workspaces/mcu-ros/src/install/yasmin/share/yasmin/local_setup.bash
$ 
```
I am on ros2 jazzy (so an ubuntu 24.04).

Thanks for your help in advance!

mgonzs13, 2024-10-09:
Hey @mowoe, thanks for the issue. This seems to be a problem due to the C++ version. I have fixed it in https://github.com/uleroboticsgroup/yasmin/commit/fbeb835008c4ab686f34c1f471bbf8ca0a2fe9e1 by including the stdexcept. You can pull it and try again.

mowoe, 2024-10-09:
Thank you so much for the lightning fast response @mgonzs13 !
It works now!

closed 2024-10-09 by mowoe