uleroboticsgroup/yasmin #26: [Viewer] How to zoom or show only a submachine?

E-Mans-Application, 2024-10-15:
Hi,

I'm giving a try to Yasmin, and I want to see a complex state machine (with nested submachines) using the viewer.

The machine is scaled to fit in the window, but that makes it too small to be readable. If I check "Hide Nested FSMs", the names of the submachines become readable, but their substates are hidden.

I need to zoom to specific areas of the machine. When I click on a state, it changes color but does not expand.
Can you provide a way to zoom in and navigate in the graph (e.g. with the mouse)?

I know there is a dropdown that allows to show only one machine, but submachines do not automatically appear in the dropdown list. Only the root machine given to YasminViewerPub appears.

If it is too hard to implement zoom, could you make submachines be added to the dropdown list automatically?

mgonzs13, 2024-10-16:
Hi @E-Mans-Application, thanks for the issue. Let me think about this, I'll return when I have some advances.

reinzor, 2024-10-21:
We are also encountering this issue.

reinzor, 2024-10-21:
Thanks for this library btw!

mgonzs13, 2024-10-27:
Zoom feature added to the viewer. Here you have an example.

https://github.com/user-attachments/assets/f28fc372-8d07-4840-9054-ef1fa0d04e22

reinzor, 2024-10-28:
Great!

reinzor, 2024-11-05:
I think we can close this issue.

closed 2024-11-05 by mgonzs13