uleroboticsgroup/yasmin #29: No state machine validation on startup

reinzor, 2024-10-21:
First of all thanks for all efforts implementing / maintaining this library :).

Coming from `SMACH`, the state machine set-up is validated on startup. When using Yasmin C++, the state machine is not validated. So the user is not notified on startup but instead sees that the state machine visualization is not working, as mentioned in https://github.com/uleroboticsgroup/yasmin/issues/28

mgonzs13, 2024-10-21:
Hi @reinzor, thanks for the issue. I'll try to push this for the next update.

reinzor, 2024-10-22:
Great! thanks!

mgonzs13, 2024-10-27:
@reinzor I have created a validate function to check the state machines https://github.com/uleroboticsgroup/yasmin/commit/2cb1a088e106bb6316ff637c92db973562a81ac7. It is for the Python version but I'll include a C++ after refining this one. The validation is based on the SMACH [check_consistency](https://github.com/ros/executive_smach/blob/7fb264e1d50c29fc03be082b9c33537eea41c5ca/smach/smach/state_machine.py#L461) although I don't know if raising an Exception with all errors is correct instead of only one error at once. What do you think?

reinzor, 2024-10-28:
Nice work; I think raising an one exception at once would already be enough. A user is going to address these one by one anyway.

mgonzs13, 2024-10-30:
Rising only one exception at once 4f90c8990d335ed39b35d5d88019d95a0a80e05d and C++ validation 657d3fe764e456007c8428b5d7c074faad7533f2a have been added.

reinzor, 2024-10-30:
Great, don't we want to call the validation function on `execute`?

mgonzs13, 2024-10-30:
Yep, I've just forgotten it 😅 and it was included in the next commit.

reinzor, 2024-10-31:
This can be closed then.

reinzor, 2024-11-04:
@mgonzs13 , I have been testing with the state machine validations and I find this validation a bit strange:

https://github.com/uleroboticsgroup/yasmin/blob/862394e3aee35c0fc514a8e4cae67f8b733bcfdb/yasmin/src/yasmin/state_machine.cpp#L214

I am not sure whether we want to enforce this. 

I am now running into problems with my setup where I have multiple toplevel state machines that have the same outcomes. These toplevel state machines are all connected but the implementation per toplevel state machine can be different. For one of these state machines I do not have a state that triggers a specific terminal state in the statemachine but the state machine should still be valid in my opinion. What are your thoughts?

reinzor, 2024-11-04:
Same holds for https://github.com/uleroboticsgroup/yasmin/blob/862394e3aee35c0fc514a8e4cae67f8b733bcfdb/yasmin/src/yasmin/state_machine.cpp#L247

reinzor, 2024-11-04:
I also see that we are calling `validate` on every sub state machine execute. I also do not think this is very desirable since then we are continuously validating, also on runtime. Not only on startup. What do you think? We can also let the user manually call `validate`.

mgonzs13, 2024-11-04:
My bad, I didn't realize that. We can use a private flag attribute to indicate if the state machine has been validated. It will be true when the state machine configuration changes that is adding states.

mgonzs13, 2024-11-04:
> @mgonzs13 , I have been testing with the state machine validations and I find this validation a bit strange: [...]

> Same holds for
> 
> https://github.com/uleroboticsgroup/yasmin/blob/862394e3aee35c0fc514a8e4cae67f8b733bcfdb/yasmin/src/yasmin/state_machine.cpp#L247

My idea with those checks was to validate that all outcomes, from states and state machines, were reached by using the transitions so that there are no unused or unecessary outcomes.

mgonzs13, 2024-11-04:
3ed8acefc05eefff9b28464b8133ffab6230d05e I've added a `strict_mode` param with a default value of `false` so that those checks are not used in a regular validation but a used can run a strict validation.

reinzor, 2024-11-05:
Great, I verified this on our set-up and it works. Thanks!

closed 2024-11-05 by reinzor