uleroboticsgroup/yasmin #30: Bloom release

reinzor, 2024-10-21:
Hi, thanks for this library again. Would it be possible to [bloom this package](https://docs.ros.org/en/jazzy/How-To-Guides/Releasing/Releasing-a-Package.html) so that it is available in the default ROS2 `Jazzy` distribution?

mgonzs13, 2024-10-21:
This was in my TODO. I may include it for this next version.

mgonzs13, 2024-11-16:
I have started the releasing process https://github.com/ros/rosdistro/pull/43524.

mgonzs13, 2024-12-06:
https://github.com/ros/rosdistro/pull/43524 merged. Creation of release team and repository started https://github.com/ros2-gbp/ros2-gbp-github-org/issues/670

mgonzs13, 2024-12-16:
PRs for bloom releases created https://github.com/ros/rosdistro/pull/43924, https://github.com/ros/rosdistro/pull/43925, https://github.com/ros/rosdistro/pull/43926

mgonzs13, 2024-12-16:
@reinzor all PRs have been merged.

closed 2024-12-17 by mgonzs13