uleroboticsgroup/yasmin #31: Log transitions using ROS

reinzor, 2024-10-24:
Hi,

Currently, the LOG implementation is using `printf`: https://github.com/uleroboticsgroup/yasmin/blob/main/yasmin/include/yasmin/yasmin_logs.hpp

Shouldn't we change this package to use ROS logging, e.g.:

```c++
RCLCPP_DEBUG(rclcpp::get_logger("yasmin"), "hi")
```

Or if this is not desired, we should expose methods for registering transition callbacks so that we can implement the logging using callback functions.

What do you think?

mgonzs13, 2024-10-24:
@reinzor The idea behind using YASMIN_LOG instead of ROS 2 was to have the yasmin package independent from ROS. How do you propose implementing the logging using callback functions?

reinzor, 2024-10-25:
I understand the idea but since we are already depending on `rclcpp`, we can also create loggers without having an active node. This way we can avoid ROS communications. The seconds option is to remove the logging completely from the library and add callbacks function that allows a user to implement logging. Example https://github.com/ros/executive_smach/blob/7fb264e1d50c29fc03be082b9c33537eea41c5ca/smach/smach/container.py#L123 

This can be done in `c++` using `std::function`

mgonzs13, 2024-10-25:
The yasmin package should not depend on rclcpp and rclpy, I have to remove those dependencies. Let me think on the callbacks solution.

mgonzs13, 2024-10-28:
@reinzor what about changing the logs similarly to `SMACH` that uses a [set_loggers](https://github.com/ros/executive_smach/blob/7fb264e1d50c29fc03be082b9c33537eea41c5ca/smach/smach/log.py#L19) function to change the loggers? This can be the resulting Python code https://github.com/uleroboticsgroup/yasmin/commit/206e69efda76120311004d2e333bfbb52334fb3a.

reinzor, 2024-10-28:
That could be an option but how are we going to do this for C++?

mgonzs13, 2024-10-28:
An idea I've also been working on is this. I may change the part of the code that sets the new loggers. What do you think?

```cpp
#ifndef YASMIN_LOGS_HPP
#define YASMIN_LOGS_HPP

#include <cstdarg>
#include <cstdio>

namespace yasmin {

// define raw function pointer type for logging
typedef void (*LogFunction)(const char *, ...);

// declare function pointers as extern
extern LogFunction log_error;
extern LogFunction log_warn;
extern LogFunction log_info;
extern LogFunction log_debug;

// macros to use the function pointers for logging
#define YASMIN_LOG_ERROR(text, ...) yasmin::log_error(text, ##__VA_ARGS__)
#define YASMIN_LOG_WARN(text, ...) yasmin::log_warn(text, ##__VA_ARGS__)
#define YASMIN_LOG_INFO(text, ...) yasmin::log_info(text, ##__VA_ARGS__)
#define YASMIN_LOG_DEBUG(text, ...) yasmin::log_debug(text, ##__VA_ARGS__)

// function to set custom log functions
void set_loggers(LogFunction info, LogFunction warn, LogFunction debug,
                 LogFunction error);

} // namespace yasmin

#endif // YASMIN_LOGS_HPP
```

```cpp
#include "yasmin/yasmin_logs.hpp"

namespace yasmin {

// define the default log functions
void default_log_error(const char *text, ...) {
  va_list args;
  va_start(args, text);
  fprintf(stderr, "[ERROR] ");
  vfprintf(stderr, text, args);
  fprintf(stderr, "\n");
  va_end(args);
}

void default_log_warn(const char *text, ...) {
  va_list args;
  va_start(args, text);
  fprintf(stderr, "[WARN] ");
  vfprintf(stderr, text, args);
  fprintf(stderr, "\n");
  va_end(args);
}

void default_log_info(const char *text, ...) {
  va_list args;
  va_start(args, text);
  fprintf(stderr, "[INFO] ");
  vfprintf(stderr, text, args);
  fprintf(stderr, "\n");
  va_end(args);
}

void default_log_debug(const char *text, ...) {
  va_list args;
  va_start(args, text);
  fprintf(stderr, "[DEBUG] ");
  vfprintf(stderr, text, args);
  fprintf(stderr, "\n");
  va_end(args);
}

// define the function pointers to be initialized with default log functions
LogFunction log_error = default_log_error;
LogFunction log_warn = default_log_warn;
LogFunction log_info = default_log_info;
LogFunction log_debug = default_log_debug;

// implement the set_loggers function
void set_loggers(LogFunction info, LogFunction warn, LogFunction debug,
                 LogFunction error) {
  log_info = info ? info : default_log_info;
  log_warn = warn ? warn : default_log_warn;
  log_debug = debug ? debug : default_log_debug;
  log_error = error ? error : default_log_error;
}

} // namespace yasmin
```

```cpp
#include <iomanip>
#include <random>
#include <sstream>

#include "yasmin/yasmin_logs.hpp"
#include "yasmin_ros/yasmin_node.hpp"

using namespace yasmin_ros;

std::string generateUUID() {
  std::random_device rd;
  std::mt19937 gen(rd());
  std::uniform_int_distribution<> dis(0, 15);
  auto rand_hex_digit = [&gen, &dis]() {
    constexpr char hex_digits[] = "0123456789abcdef";
    return hex_digits[dis(gen)];
  };

  std::stringstream ss;
  for (int i = 0; i < 32; ++i) {
    if (i == 8 || i == 12 || i == 16 || i == 20)
      ss << "_";
    ss << rand_hex_digit();
  }
  return ss.str();
}

#include <cstdarg>
#include <rclcpp/rclcpp.hpp>
#include <string>

void ros_log_error(const char *text, ...) {
  va_list args;
  va_start(args, text);

  int size = vsnprintf(nullptr, 0, text, args) + 1; // +1 for '\0'
  va_end(args);

  std::string buffer(size, '\0');

  va_start(args, text);
  vsnprintf(&buffer[0], buffer.size(), text, args);
  va_end(args);

  RCLCPP_ERROR(YasminNode::get_instance()->get_logger(), "%s", buffer.c_str());
}

void ros_log_warn(const char *text, ...) {
  va_list args;
  va_start(args, text);

  int size = vsnprintf(nullptr, 0, text, args) + 1;
  va_end(args);

  std::string buffer(size, '\0');

  va_start(args, text);
  vsnprintf(&buffer[0], buffer.size(), text, args);
  va_end(args);

  RCLCPP_WARN(YasminNode::get_instance()->get_logger(), "%s", buffer.c_str());
}

void ros_log_info(const char *text, ...) {
  va_list args;
  va_start(args, text);

  int size = vsnprintf(nullptr, 0, text, args) + 1;
  va_end(args);

  std::string buffer(size, '\0');

  va_start(args, text);
  vsnprintf(&buffer[0], buffer.size(), text, args);
  va_end(args);

  RCLCPP_INFO(YasminNode::get_instance()->get_logger(), "%s", buffer.c_str());
}

void ros_log_debug(const char *text, ...) {
  va_list args;
  va_start(args, text);

  int size = vsnprintf(nullptr, 0, text, args) + 1;
  va_end(args);

  std::string buffer(size, '\0');

  va_start(args, text);
  vsnprintf(&buffer[0], buffer.size(), text, args);
  va_end(args);

  RCLCPP_DEBUG(YasminNode::get_instance()->get_logger(), "%s", buffer.c_str());
}

YasminNode::YasminNode() : rclcpp::Node("yasmin_" + generateUUID() + "_node") {

  yasmin::set_loggers(ros_log_info, ros_log_warn, ros_log_debug, ros_log_error);

  this->executor.add_node(this->get_node_base_interface());
  this->spin_thread = std::make_unique<std::thread>(
      &rclcpp::executors::MultiThreadedExecutor::spin, &this->executor);
  this->spin_thread->detach();
}
```

reinzor, 2024-10-28:
I am not sure whether we should put this functionality in the `YasminNode`. We are for example not using anything from `yasmin_ros` and also not the `YasminNode` because we want to have control over our threads and executors. If we can expose this as a separate convenience function, that would be better I think.

mgonzs13, 2024-10-28:
Another option is to create a ROS 2 logger, for instance

```cpp
void ros_log_info(const char *text, ...) {
  va_list args;
  va_start(args, text);

  int size = vsnprintf(nullptr, 0, text, args) + 1;
  va_end(args);

  std::string buffer(size, '\0');

  va_start(args, text);
  vsnprintf(&buffer[0], buffer.size(), text, args);
  va_end(args);

  RCLCPP_INFO(rclcpp::get_logger("yasmin_logger"), "%s", buffer.c_str());
}
```

reinzor, 2024-10-28:
That would make more sense I think. And if we can provide a convenience function for registering these, that would be great.

mgonzs13, 2024-10-28:
Same for Python. However, these logs are not published in the `/rosout'` topics.

```python
import rclpy.logging
rclpy.logging.get_logger("yasmin_logger").info("my message")
```

reinzor, 2024-10-28:
I did not think about this but this is indeed true. I do think that this is desirable if you want to inspect the logging remotely. Maybe removing the logging from the library and create the functionality to register transition callbacks isn't a bad option after all. Then the use will have all flexibility.

reinzor, 2024-10-28:
Something like would then already do the trick.

```
StateMachine sm;
sm.register_transition_callback([node](auto from, auto to) {
  RCLCPP_INFO(node->get_logger(), "%s --> %s", from.c_str(0, to.c_str());
});
```

mgonzs13, 2024-10-28:
The callback feature seems quite interesting to me but the problem is that you have to register the same callback each time you create a new state machine, right?

reinzor, 2024-10-29:
Yes, the callback functions need to be installed recursively.

mgonzs13, 2024-10-31:
I've added the callbacks e98f5f8094899b2424755cb35710a30e36fa806a but I'm still maintaining the logs, I've added more logs 6d56dff794b346d9d15ef2eb998ee094e8987cc1. What more info do you consider important to log?

reinzor, 2024-10-31:
Callbacks are a great addition! This also enables us to publish state updates on change, e.g. for the viewer.

Regarding the logs, I still think it is desired to log everything via ROS logging on `/rosout`. Can't we make the library dependent on `rclcpp` and `rclpy` for logging only? I think this is far the easies solution. We can by default use a logger that does not dependend on the node but allow passing a logger in the state machine constructor from the node, e.g.

```
StateMachine(std::vector<std::string> outcomes, rclcpp::Logger logger= rclcpp::get_logger("yasmin"))
```

This is a ROS package after all and `rclcpp`,`rclpy` will come with all ros installations.

reinzor, 2024-10-31:
Ah great, thanks!

reinzor, 2024-10-31:
I think we can close this one then.