uleroboticsgroup/yasmin #32: Improve logging statements

reinzor, 2024-10-24:
We only log here:

https://github.com/uleroboticsgroup/yasmin/blob/d9c84298dcebc92dccbaad99f519bdaf84ad6515/yasmin/src/yasmin/state_machine.cpp#L113

Therefore, we are not logging the start state of the state machine. 

Besides this, it is also not really clear if we are entering a child state machine which makes it sometimes hard to follow the state machine flow from logs. 

Another idea would also be to log the full state path when using hierarchical state machines, something like:

```c++
auto get_current_state_path(const std::shared_ptr<yasmin::State> & state, std::string path = "")
{
  const auto state_machine = std::dynamic_pointer_cast<yasmin::StateMachine>(state);
  if (state_machine == nullptr) {
    return path;
  }

  const auto current_state_name = state_machine->get_current_state();
  const auto current_state_it = state_machine->get_states().find(current_state_name);
  if (current_state_it == state_machine->get_states().end()) {
    return path;
  }

  return get_current_state_path(current_state_it->second, path + "/" + current_state_name);
}
```

What do you think?

mgonzs13, 2024-10-24:
@reinzor You are right about the start sate. The ```get_current_state_path``` function ide seems great to me.

reinzor, 2024-10-25:
Implementing better logging would only make sense if we agree to put logging into the core of the library, outcome of https://github.com/uleroboticsgroup/yasmin/issues/31

mgonzs13, 2024-11-04:
@reinzor I have been added logs for both Python and C++. What more info do you consider to log?

reinzor, 2024-11-05:
For now, it seems ok I guess, thanks! Closing this ticket.

closed 2024-11-05 by reinzor