uleroboticsgroup/yasmin #33: yasmin_viewer missing dependencies

reinzor, 2024-10-24:
```
[yasmin_viewer_node-48]   File "/install/yasmin_viewer/lib/yasmin_viewer/yasmin_viewer_node", line 35, in <module>
[yasmin_viewer_node-48]     from waitress import serve
[yasmin_viewer_node-48] ModuleNotFoundError: No module named 'waitress'
```
The `waitress` dependency is not listed in the [`package.xml`](https://github.com/uleroboticsgroup/yasmin/blob/main/yasmin_viewer/package.xml)

It can be resolved using rosdep: https://github.com/ros/rosdistro/blob/faf08fa8c6a18790b52e984ab9a2d643911a785d/rosdep/python.yaml#L10283

reinzor, 2024-10-24:
Same holds for `expiringdict`:

```
Traceback (most recent call last):
  File "/install/yasmin_viewer/lib/yasmin_viewer/yasmin_viewer_node", line 36, in <module>
    from expiringdict import ExpiringDict
ModuleNotFoundError: No module named 'expiringdict'
```

reinzor, 2024-10-24:
> Same holds for expiringdict

Added that one here https://github.com/ros/rosdistro/pull/43261

reinzor, 2024-10-24:
And `flask`: https://github.com/ros/rosdistro/blob/faf08fa8c6a18790b52e984ab9a2d643911a785d/rosdep/python.yaml#L6124

mgonzs13, 2024-10-24:
Nice work @reinzor . I really like depending only on rosdep for the dependencies. I'll have to check more on the available python packages.

reinzor, 2024-11-05:
This issue can be closed if:
- we remove the `requirements.txt`
- Update the `README.md` installation instructions
- https://github.com/ros/rosdistro/pull/43261 is merged

reinzor, 2024-11-06:
rosdep rule is merged

closed 2024-11-13 by mgonzs13