uleroboticsgroup/yasmin #40: State name the same as outcome

reinzor, 2024-11-08:
I ran into an issue:

[image]

So in code we are transitioning from the `succeeded` state to the `succeeded` outcome. But the viewer shows something different. I think also this state machine is not correct but it is accepted by the framework. I think it should reject since the behavior is undefined.

mgonzs13, 2024-11-08:
@reinzor I'll check the viewer. What do you mean with `this state machine is not correct`?

reinzor, 2024-11-08:
Because how can the transition know it is for the state machine `succeeded` or for the state `succeeded`. They should not have the same name, that shouldn't be allowed I think?

mgonzs13, 2024-11-08:
Yep, I didn't realize. I'll add this check to the validation in the state machine.

mgonzs13, 2024-11-08:
53ffd21 checking if the state name is an outcome when adding states should be enough, right?

reinzor, 2024-11-12:
Should be sufficient yes, great!

closed 2024-11-12 by reinzor