uleroboticsgroup/yasmin #42: License issue

Rayman, 2025-01-27:
Hi,

Thank you for creating this package. We would like to use yasmin in a proprietary product, but we are unable to do this with the current license. When linking against GPL-v3 code, the license forces us to re-release also under GPL-v3, which would force us to open-source our code.

Is it possible to release under a different license or add a linking exception?  We would prefer Apache-2, BSD or MIT license, but I think LGPL would also be fine.

Thank you for your response

mgonzs13, 2025-01-28:
Hi @Rayman, thanks for your interest in this project. We are discussing the future of this repo including changing the license since several people have asked for it. It may change in the next releases.

PaulVerhoeckx, 2025-02-17:
Hi,

I’m running into a similar issue as @Rayman and was wondering if you’ve been able to reach a conclusion on this.

We’d love to contribute and collaborate on this repo by submitting PRs, but if using it in proprietary products isn’t a viable option, we’ll have to explore other solutions. Looking forward to your thoughts!

mgonzs13, 2025-03-04:
I'm sorry for not getting back to you sooner, but I'm having some issues with this. I'll come back asap.

PaulVerhoeckx, 2025-03-31:
@mgonzs13, Any update on this?

MCFurry, 2025-04-04:
I'd really like to use this library as well, but such a license is preventing me from doing so. @mgonzs13 could you please reconsider this?
IMHO it would be a shame if such a cool framework is not being used more often!

johnsonTyler0511, 2025-05-06:
I am also interested in using it and just want to hop on for more support.

mgonzs13, 2025-05-27:
> @mgonzs13, Can we conclude it remains GPL-v3? As there has been several (minor) releases since the issue was created..

Yes, for now, the project will remain under GPL-v3 — at least temporarily. I haven’t had the time to properly address the licensing discussion with the rest of the team, and I apologize for the delay. Realistically, the most likely outcome is that we’ll continue with GPL-v3, but I’ll make sure to share any updates if that changes.

closed 2025-07-11 by github-actions[bot]