uleroboticsgroup/yasmin #43: Call a StateMachine from a StateMachine

gadorneles, 2025-02-13:
Hello, thanks for all your work.

I'd like to divide my FSM in a few state machines that I can call from a higher level state machine, like we used to do with SMACH and ROS.

However, the StateMachine class doesn't seem to be able to receive StateMachine as a type in the add_state method. What is the preferred way to do what I intend? Would I need to work with the State class only?

edompedro, 2025-02-13:
I'm having the same problem. With this approach the higher level state machine became simple and the sub machines more complex.

mgonzs13, 2025-02-13:
Hi @gadorneles, since StateMachine inherits from State, you can use any StateMachine as a State, including adding it to another StateMachine. Sorry if this is not clear in the docs. Here you have an [example](https://github.com/uleroboticsgroup/yasmin/blob/main/yasmin_demos/yasmin_demos/nav_demo.py):

```python
    # Create state machines
    sm = StateMachine(outcomes=[SUCCEED, ABORT, CANCEL])
    nav_sm = StateMachine(outcomes=[SUCCEED, ABORT, CANCEL])

    # Add states to the state machine
    sm.add_state(
        "CREATING_WAYPOINTS",
        CbState([SUCCEED], create_waypoints),
        transitions={
            SUCCEED: "TAKING_RANDOM_WAYPOINTS",
        },
    )
    sm.add_state(
        "TAKING_RANDOM_WAYPOINTS",
        CbState([SUCCEED], take_random_waypoint),
        transitions={
            SUCCEED: "NAVIGATING",
        },
    )

    nav_sm.add_state(
        "GETTING_NEXT_WAYPOINT",
        CbState([END, HAS_NEXT], get_next_waypoint),
        transitions={
            END: SUCCEED,
            HAS_NEXT: "NAVIGATING",
        },
    )
    nav_sm.add_state(
        "NAVIGATING",
        Nav2State(),
        transitions={
            SUCCEED: "GETTING_NEXT_WAYPOINT",
            CANCEL: CANCEL,
            ABORT: ABORT,
        },
    )

    sm.add_state(
        "NAVIGATING",
        nav_sm,
        transitions={
            SUCCEED: SUCCEED,
            CANCEL: CANCEL,
            ABORT: ABORT,
        },
    )
```

@edompedro I hope this can help you too.

gadorneles, 2025-02-13:
Thanks for the quick response!

closed 2025-02-13 by gadorneles