uleroboticsgroup/yasmin #52: C++ state call from other file

edompedro, 2025-04-02:
Hii, is there a way to declare a C++ State in one file and calling this state in another file? Assuming that the main file coordinate many States and to be more clear this states are declare and implement in separated files.
I actually searching fr some example when a State was declare outside the main use file, but unfortunately I could not find any example related to this.

mgonzs13, 2025-04-04:
@edompedro Yes, you can separate each C++ state into different files, just like you would in standard C++ development. While there isn’t a specific example available, you can create a header (.hpp) and source (.cpp) file for each state, similar to how State or ServiceState is structured. Once created, you can include and use these new states as needed.

edompedro, 2025-04-07:
Sorry for take a long time to view the response. Thanks, I will try to do this with an example and I can add to the community.

closed 2025-04-07 by edompedro