uleroboticsgroup/yasmin #60: Issue using a YasminNode between multiple states.

gadorneles, 2025-06-16:
I have two states in sequence, one of them calls a service and the other calls an action. Both are using  YasminNode.get_instance() as the node to call the service and the action. Whenever I run my code, the service call runs smoothly but the action is stuck. 

Situations where it works:

- If I switch the states to utilize two different rclpy.Node objects.
- If I run them independent of one another.

Any ideas what could be causing this problem? Shouldn't the MultiThreadExecutor be able to handle this?

mgonzs13, 2025-06-16:
Hi @gadorneles, you shouldn't use spin with the YasminNode since it has its own spin. Why not use the ServiceState?

gadorneles, 2025-06-16:
Hello! Thanks for the quick response. Removing the spin did fix my issue. Do you have an idea as to why another spin would completely break the spin running in the other thread?

gadorneles, 2025-06-20:
I understand, thanks for the answers!

closed 2025-06-20 by gadorneles